rgleason/vdr_pi #15: Master-Rasbats Branch: need full build & deployment support for all supported environments

rgleason, 2021-07-17:
The new Icons are working adapting Rasbats version of VDR_pi.
Now I am having issues getting full build and deployment of all build environments.
Before I can push this up to plugins I need to get them all working.
The git repository is  [rgleason/vdr_pi     Branch:  master-rasbats
](https://github.com/rgleason/vdr_pi/tree/master-rasbats)
We are missing or they fail:
 [Here are the failures](https://app.circleci.com/pipelines/github/rgleason/vdr_pi?branch=master-rasbats)

1.  [Macos failure ](https://app.circleci.com/pipelines/github/rgleason/vdr_pi/155/workflows/a7ed1339-0b90-4460-b73b-71f02ad01047/jobs/1252)- I've changed to the more current xcode  12.5.1  but it still does not complete.
2.  Raspian - both versions do not show up and are missing 
3. [Flatpak  18.04  Failure](https://app.circleci.com/pipelines/github/rgleason/vdr_pi/155/workflows/a7ed1339-0b90-4460-b73b-71f02ad01047/jobs/1252)
4.  I thought there was supposed to be a  Flatpak version.

Because this is am important plugin for testing we should have this working for all environments.
Once working I will make a PR to bdbcat's repository.

rgleason, 2021-07-18:
This branch  ["master-rasbats](https://github.com/rgleason/vdr_pi/tree/master-rasbats)" is now working with the svg icons, and building for all environments except 
Flatpak 18.08  and MacOS.
I assume we are not supposed to build Flatpak 18.08.
I would like to get the MacOS build fixed.
https://app.circleci.com/pipelines/github/rgleason/vdr_pi/159/workflows/ff4d2355-13cd-4147-ac6e-998b0d71661c/jobs/1295

```
[ 92%] create-pkg [vdr_pi]: Generating pkg file.
==============================================================================
ERROR:
Description:
Unable to copy item at path '/Users/distiller/project/build/Resources' to '/private/tmp/WMjbSHTg/502/VDR' because the item could not be found
Step:
Project > Distribution > Package 'VDR-Plugin' > Payload > Assembl

```
Any ideas how to fix this?

rgleason, 2021-07-18:
Here is how it looks right now.  Master-Rasbats branch

[image]

Rasbats, 2021-07-19:
Try changing Vdr_pi.pkgproj.in to VDR_pi.pkgproj.in, in the buildosx folder.

closed 2021-07-29 by rgleason