rgleason/vdr_pi #28: overwrite dialog

rgleason, 2022-09-06:
The VDR plugin requires a fix: if a file already exists, it should have the standard dialog that first asks if you wish to overwrite an existing file.

rgleason, 2022-09-09:
https://www.cruisersforum.com/forums/f134/vdr-plugin-save-file-fix-needed-268392.html#post3677148

This is a "bad good" idea...  Indeed, this solution multiplies the clicks and this is not practical in a cockpit, in navigation, especially if you also have to write the name of the route.  Personally, I am asking for a simplification that I had requested in 2019 in the tracker FS#2592 - simplify the backup process. Attached is the copy/paste of the request:  

Personally, I am asking for a simplification that I had requested in 2019 in the tracker FS#2592 - simplify the backup process.
Attached is the copy/paste of the request:

```
**FS#2592 - simplify the backup process**

To save the [NMEA](https://www.cruisersforum.com/forums/tags/nmea.html) data with VDR, you have to click on the icon, a window opens and you have to write a name for this recording that has been started.
This is long and especially not practical when navigating...

Would it be possible to automate this process? (a bit like the logbook)

So, for example, the process could be the following:
- we click on the icon "VDR recording"
- automatically, a file is created in ProgramData/opencpn/plugins/VDR/******. txt

where the file name *****. txt would be of the style: VDR_Year / month / day / hour / minute / second.
Example: VDR_20190704164533.txt (4 July 2019 at 16h45mn33sec).

```
Knowing that, in addition to the date, the time to the nearest minute (or even second) will be indicated, there will be no risk of overwriting existing data. and, in addition, the [registration](https://www.cruisersforum.com/forums/tags/registration.html) process is really simplified...

TwoCanPlugIn, 2022-09-15:
Some suggestions for file naming/saving. These are used in the TwoCan plugin for saving log files.

1. Save the file in the user's documents folder.  Easily accomplished with wxWidgets using the wxStandardPaths::GetDocumentsDir() API. 
ProgramData, Program FIles and AppData are not the proper places to store users' data files. 

2. Automagically generate file names, using a timestamp, eliminating the file overwriting problem.
wxDateTime tm = wxDateTime::Now();
wxString fileName = tm.Format("vdr-%Y-%m-%d_%H%M%S.txt");

rgleason, 2022-09-15:
Thanks, will look at twocan_pi

TwoCanPlugIn, 2022-09-16:
Here's a link to the actual code. Saves you from having to peruse lots of unnecessary stuff
https://github.com/TwoCanPlugIn/TwoCanPlugIn/blob/master/src/twocandevice.cpp#L95
Regards.

rgleason, 2022-09-16:
Thanks, Line 94

```
// Any raw logging ?
	if (logLevel > FLAGS_LOG_NONE) {
		wxDateTime tm = wxDateTime::Now();
		// construct a filename with the following format twocan-2018-12-31_210735.log
		wxString fileName = tm.Format("twocan-%Y-%m-%d_%H%M%S.log");
		if (rawLogFile.Open(wxString::Format("%s//%s", wxStandardPaths::Get().GetDocumentsDir(), fileName), wxFile::write)) {
			wxLogMessage(_T("TwoCan Device, Created log file: %s"), fileName);
			// If a CSV format initialize with a header row
			if (logLevel == FLAGS_LOG_CSV) {
				rawLogFile.Write("Source,Destination,PGN,Priority,D1,D2,D3,D4,D5,D6,D7,D8\r\n");
			}
		}
		else {
			wxLogError(_T("TwoCan Device, Unable to create raw log file: %s"), fileName);
		}
	}
}
```

rgleason, 2022-09-18:
One other user opinions:

> My habits are different.
> 
> I created a [destination](https://www.cruisersforum.com/forums/tags/destination.html) folder of all my VDR files. And since the [current](https://www.cruisersforum.com/forums/tags/current.html) "official" version of VDR always remembers the last folder in which the user stored the last saved file, saving a new folder goes there automatically. Let's keep that intact.
> 
> So let VDR use the last folder where VDR should save the file.
> 
> Personally, I change this folder once a year, during my first annual [navigation](https://www.cruisersforum.com/forums/tags/navigation.html). And [current](https://www.cruisersforum.com/forums/tags/current.html)ly, the method to change this folder is fine. So, I persist and sign, let's leave it as it is.
> 
> 
> Concerning the name of the file :
> The idea of ​​starting with the date by putting the year, then the month, then the day, automatically, is good. Sorting is done nicely. This is what I do manually with the current official version of VDR.
> 
> Just one detail, I find it more convenient to separate the three date fields with dashes of the 6. (i.e. 2022-09-17 is easier to read than 20220917).
> 
> But, personally, I also add the day. So for today it would be "2022-09-17_Saturday". But that still doesn't satisfy me. I always add the departure port and the arrival port. For example "2022-09-17-Saturday_Paimpol_Binic".
> 
> So, let me decide when, I'm happy with the file name. So, let me validate the file name.

rgleason, 2023-08-17:
Example file name
2022-09-17-Sat-07-54-03-Paimpol_Binic

Click on the icon "VDR recording"
Automatically, a file name is created in the folder location previously selected by the user using the browser button.
where the file name *****. txt would be of the style:  "Year / month / day / hour / minute / second / [Paused for user input]"

Example: 2019-07-06-[Weekday]-16-45-33[-user input].txt

1. User allowed to select destination folder with the file browser.
2. Create the first part of the filename automatically  "2022-09-17-Wed-21-32-59[-User Input]"   
3. "year-month-day-weekday-HMS-user input"
4. Pause for user to add a qualifier (depart and arrival).
5. User hits "Enter" to complete the file name and start logging.

// Any raw logging ?  [Twocan Link](https://github.com/TwoCanPlugIn/TwoCanPlugIn/blob/a5b5ff3944a1e9bc6a5f3d3e07c4eeae3232ba3b/src/twocandevice.cpp#L95)
	if (logLevel > FLAGS_LOG_NONE) {
		wxDateTime tm = wxDateTime::Now();
		// construct a filename with the following format vdr-2018-12-31_%H%M%S_weekday.txt
		wxString fileName = tm.Format("%Y-%m-%d_%H%M%S.log");
		if (rawLogFile.Open(wxString::Format("%s//%s", wxStandardPaths::Get().GetDocumentsDir(), fileName), wxFile::write)) {
			wxLogMessage(_T("VDR Created txt file: %s"), fileName);
			// If a CSV format initialize with a header row
			if (logLevel == FLAGS_LOG_CSV) {
				rawLogFile.Write("Source,Destination,PGN,Priority,D1,D2,D3,D4,D5,D6,D7,D8\r\n");
			}
		}
		else {
			wxLogError(_T("VDR unable to create raw log file: %s"), fileName);
		}
	}
}

I think the vdr_pi.cpp    https://github.com/rgleason/vdr_pi/blob/filecreate/src/vdr_pi.cpp
is where the changes would be made.

rgleason, 2023-08-17:
After looking at this a little further, I think what we need is pretty simple.
When Record is hit, a "Create a file to record" file browser pops up.
And the first part of the file name is filled in with
year-month-day-XXX weekday (3 digits)-Hr-Min-Sec-

[image]

rgleason, 2023-08-18:
https://www.cruisersforum.com/forums/f134/autostart-plugins-268646.html

sebastien-rosset, 2024-12-11:
I've added VDR preferences and controls in #39 which hopefully addresses this request.
When the user clicks the "Record" button, it immediately starts recording without asking for a file name. There is a default output directory, and the file names are generated. The user can change the output directory in the plugin preferences.

sebastien-rosset, 2024-12-16:
The VDR plugin writes a new file when it starts recording, so there is no longer a need to pop-up a dialog box. I'm sure we can do further enhancements. Perhaps this issue can be closed, and other enhancements can be tracked in a separate issue?

rgleason, 2024-12-19:
Thank you Sebastien! Closing

closed 2024-12-19 by rgleason