rgleason/vdr_pi #33: RFE VDR Replay improvements

rgleason, 2023-08-17:
1. Ability to select a start point by having a slider to locate the start.
2. Ability to Start, Stop, Pause, Continue
3. In addition to the current slider for speed.

sebastien-rosset, 2024-12-04:
It appears (though I'm not entirely sure) the replay speed can jump or slow down, depending on how much data is present in the VDR file. The replay speed is not based on reading timestamps. RFE #37 could potentially help.

sebastien-rosset, 2024-12-04:
When replaying a VDR, IMHO it's more useful to show the GPS time versus the computer clock at the time the VDR is replayed. In the dashboard we can show both, which is useful. However, since there is a GPS watchdog timer, the GPS clock in the dashboard would be invalidated instead of "paused". See https://github.com/OpenCPN/OpenCPN/pull/4083

Also, in the status bar plugin, the `%d` and `%t` directives use the current computer clock, which is completely different from the GPS time when replaying VDR files.

rgleason, 2024-12-05:
Thanks Sebastien.

sebastien-rosset, 2024-12-18:
I think this can be closed.

closed 2024-12-18 by rgleason