rgleason/vdr_pi #35: File open fails on Android

nocheinpeter, 2024-09-29:
Does no longer work with official openCPN app v 5.10.2 (111).

Pressing "play" icon open dir selection and not file selection dialogue. Then ther is no replay possible.

Affects minimum Android version 11,12,14.

rgleason, 2024-09-30:
Are you downloading and installing https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/detail/raw/VDR-v1.3.31-msvc-wx32-10-tarball/v1.3.31/   which shows as v1.3.0 when installed?

This works for me.
Rick

nocheinpeter, 2024-09-30:
It fails on Android, is this windows platform?

nocheinpeter, 2024-09-30:
I installed the above Android version, does not work.

nocheinpeter, 2024-09-30:
[opencpn (4).log](https://github.com/user-attachments/files/17184683/opencpn.4.log)

Last log attached. VDR replay and record were last actions..Both do not work

rgleason, 2024-09-30:
Are you using Android or Windows, please.
If using android, that is the wrong tarball.

the new VDR is now in the opencpn/plugins catalog.  Please update your plugin and install.
Also Please close this Issue.

> 07:53:02.377 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins/o-charts_pi
> 07:53:02.623 MESSAGE plugin_loader.cpp:443 Checking plugin candidate: /data/user/0/org.opencpn.opencpn/manPlug/libvdr_pi.so
> 07:53:02.624 MESSAGE plugin_loader.cpp:514 Checking plugin compatibility: /data/user/0/org.opencpn.opencpn/manPlug/libvdr_pi.so
> 07:53:02.624 MESSAGE plugin_loader.cpp:1396 PluginLoader: Loading PlugIn: /data/user/0/org.opencpn.opencpn/manPlug/libvdr_pi.so
> 07:53:02.632 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins;/data/user/0/org.opencpn.opencpn
> 07:53:02.632 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins/vdr_pi
> 07:53:02.632 MESSAGE icons.cpp:69 Loading toolbar icon: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins/vdr_pi/data/vdr_record_toggled.svg
> 07:53:02.632 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins;/data/user/0/org.opencpn.opencpn
> 07:53:02.633 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins/vdr_pi
> 07:53:02.676 MESSAGE plugin_loader.cpp:422 Loading plugins from dir: /data/user/0/org.opencpn.opencpn
> 07:53:02.676 MESSAGE plugin_loader.cpp:646 PluginLoader searching for PlugIns in location /data/user/0/org.opencpn.opencpn
>

nocheinpeter, 2024-09-30:
As the title says, its **Android**. On **Android** the plugin fails as described above.
I just updated catalog and re-installed VDR v1.3.31. After install it shows version number 1.3.0 at the plugin list.

Replay or recording does not work. The logfile above is the only information I have.

Pressing "play" button shows a file selection dialog. Icon shows green play button. Then I select an older recorderd file. The the play button turns back to grey and nothing happens.

Pressing "record" button shows icon red, then I cannot enter a file name but select an existing one. Then icon turns back grey and nothing happens.

rgleason, 2024-10-01:
Ok it built but there problems that need to be resolved.. Thanks. Sorry for the trouble.

rgleason, 2025-01-01:
VDR works now.  Please update the catalog and try it. 
If it works please close this.
Also see another Issue Post about 

Importing Android Plugin Tarballs  and  Download  VDR-test-files.zip

closed 2025-01-05 by rgleason