rgleason/vdr_pi #45: Record NMEA 2000 sentences

sebastien-rosset, 2024-12-14:
Ability to save NMEA 2000 sentences. For example, canboat.log format could be used.
This would require a change to the plugin API in OpenCPN.

TwoCanPlugIn, 2024-12-18:
Hi Sebastien,

Have just cobbled together some POC code for recording and replay of NMEA 2000. 

At the moment as there is no mechanism to "inject" NMEA 2000 data into core OpenCPN, the simple workaround is to simply transmit NMEA 2000 data using UDP. The user just needs to configure a NMEA 2000 UDP connection. It seemed to work.

There are a a couple of NMEA 2000 formats supported by OpenCPN over a network connection, I've just used Actisense ASCII just because it was quick to code. The format is documented [here](https://actisense.com/knowledge-base/nmea-2000/w2k-1-nmea-2000-to-wifi-gateway/nmea-2000-ascii-output-format/)

I chose not to use a serial connection as it would require users to setup a pair of virtual ports using com0com or socat and that would just make things a bit too difficult for many users.

For diagnostics, where I think the discussion should consider the role of connection debugging rather than VDR, other formats may be better, such as candump or canboat both of which can be parsed by canboat's analyzer, or Actisense EBL which can be parsed by both canboat analyzer and Actisense EBL Reader. I have my own toolsets that parse a much larger range of formats.

Anyway perhaps this attached code may help bootstrap your work.

Good luck.
[vdr-poc.cpp.txt](https://github.com/user-attachments/files/18187923/vdr-poc.cpp.txt)

closed 2024-12-18 by sebastien-rosset

sebastien-rosset, 2024-12-18:
> At the moment as there is no mechanism to "inject" NMEA 2000 data into core OpenCPN, the simple workaround is to simply transmit NMEA 2000 data using UDP. The user just needs to configure a NMEA 2000 UDP connection. It seemed to work.

Ah yes, thanks! This is exactly what I'm doing for testing purpose. I have a simulator that outputs NMEA sentences over UDP. However, I'm not sure how to make this part of a replay capability. IMO, "replay" should just work with the user having to create a UDP connection. I know it's easy to create a UDP connection in OpenCPN, but I feel it should be easier than that. The user should click "Play", and playback should start without having to manage connections.

On a related note, IMO during recording, OpenCPN should set the timestamp with millisecond precision in the event itself, rather than the plugin setting the time when it receives the message through a listener. I don't yet have hard data to show the percentile latency. Given that most sensors can send at 10Hz, sometimes 20Hz, perhaps that latency is irrelevant. I'm just experimenting with pandas and ML tools to analyze the data. Because the variables are not independent and there is a time correlation, not having to worry about timing accuracy would be a good goal.

TwoCanPlugIn, 2024-12-18:
Hi Sebastien,

I personally differentiate NMEA 2000 and NMEA 0183 using this terminology:
NMEA 2000 data is "messages" whilst NMEA 0183 data is "sentences"

When you say 
> I have a simulator that outputs NMEA sentences over UDP 

I take that to mean NMEA 0183. That can be replayed by a plugin today using the PushNMEABuffer API, which is what VDR uses. 

However at present there is no NMEA 2000 equivalent to that API. That is why the only way to "replay" logged NMEA 2000 data is to use the supported connections, noting that SocketCAN is only supported on Linux. This technique can be used to replay both existing log files and simulated voyages.

> OpenCPN should set the timestamp with millisecond precision in the event itself

IIRC only NMEA 2000 data that has originated from an Actisense NGT-1 has a "valid" timestamp. For whatever reason Dave et al chose, for all other NMEA 2000 data sources; SocketCAN, the Actisense raw text formats and I think Yacht Devices, they are  "converted" to the Actisense binary format but the timestamp and CRC checksum fields are not filled in.  Some of the reasoning for adopting the Actisense binary format internally was because the NGT-1 being a smart device, re-assembles NMEA 2000 Fast Packet messages so the data can be placed onto the message bus without any additional processing. Other data formats, notably SocketCAN are "dumb" devices, and re-assembly of Fast Messages is performed by the OpenCPN SocketCAN code before being placed onto the message bus.

sebastien-rosset, 2024-12-18:
> I personally differentiate NMEA 2000 and NMEA 0183 using this terminology:
NMEA 2000 data is "messages" whilst NMEA 0183 data is "sentences"

Sorry you are right, I took a shortcut of language.

> However at present there is no NMEA 2000 equivalent to that API. That is why the only way to "replay" logged NMEA 2000 data is to use the supported connections, noting that SocketCAN is only supported on Linux. This technique can be used to replay both existing log files and simulated voyages.

Understood. For existing OpenCPN version, yes, the VDR plugin could emit NMEA 2000 over UDP, and the user would be responsible for creating a NMEA/UDP network connection in OpenCPN, with the right port. It's a bit cumbersome to use, but it works. How about future versions of OpenCPN? Can we entertain the idea of adding APIs such that users don't have to manually create connections? There are multiple possibilities on the "how" part. For now I'm just wondering if it's something that could be considered.

> IIRC only NMEA 2000 data that has originated from an Actisense NGT-1 has a "valid" timestamp. For whatever reason Dave et al chose, for all other NMEA 2000 data sources;

Ah right. I was thinking about timestamping the data as soon as it is received by OpenCPN, rather than when it is received by the plugin. Perhaps in practice, the latency is always very low.

TwoCanPlugIn, 2024-12-19:
> How about future versions of OpenCPN? Can we entertain the idea of adding APIs such that users don't have to manually create connections? 

Well beyond my pay grade! Seems like you have excellent C++ knowledge and are quite up to speed with understanding the OpenCPN code base so perhaps you could add the feature and submit a PR. I gave up trying to understand what was then spaghetti code a long time ago.
Alternatively, submit a Request For Enhancement (RFE), squeak loudly enough, and perhaps Dave, Alec & Pavel will consider it.