rgleason/vdr_pi #52: VDR 1.3.39 imported  small bugs

rgleason, 2024-12-18:
@sebastien-rosset 

Using this tarball for Windows https://cloudsmith.io/~opencpn/repos/vdr-alpha/packages/?q=1.3.39.0+msvc+wx32+tarball

When using an old recorded vdr file  from Harmut  with the new version 1.3.39 
I find that moving the time line does not really work very well.
When I try to move the start it is not recognized.  
I will show a few screenshots and upload a zipped text file of an older "Harmut" file recording so you can see what happens.
[image]
[image]
[image]
[image]
[image]

[Hartmut-AN-Markermeer-Wind-AIS-6m.zip](https://github.com/user-attachments/files/18188721/Hartmut-AN-Markermeer-Wind-AIS-6m.zip)

Don't know if it is related to the timestamp code. 
Perhaps if it has no timestamp, then the timeline that should be used should be proportional to the size of the file?

sebastien-rosset, 2024-12-18:
It looks like there is a layout issue on your windows. The date is printed, and there is not enough space to show the timestamp. I'll take a look.

rgleason, 2024-12-18:
Yes, that is probably correct about the time.
Also, for example, if I move the time bar all the way to the right, and even click on pause and then click on start, the boat stays shown where it is, not at the end of the route.  See the second to last screenshot..   The boat seems reluctant to move.

Sebastien, these changes to the plugin are great. I am certain that many will appreciate them. Thank you.

sebastien-rosset, 2024-12-19:
Thanks for the zip file. That helps to troubleshoot. I'll probably add unit tests

> Perhaps if it has no timestamp, then the timeline that should be used should be proportional to the size of the file?

Yes, it makes sense. That was my intent, but I did not test that part thoroughly. Also, I see the file does have timestamps that are usable.

sebastien-rosset, 2024-12-19:
I've been able to reproduce the progress issue.

rgleason, 2024-12-29:
@sebastien-rosset  
I am still having issues with VDR playing older vdr created files. I have tried a half dozen files and none of them really work.
I think we need a set of testing files of various types to use to test the capabilities of the newer plugin.
I will make such an issue and post about 6 files for testing. This issue will also have a link to the older version of the plugin 1.3.31 and it will be showing the tarballs so the appropriate one can be downloaded and "Import Plugin" used to compare the results.

Also, I don't have any recorded VDR files from the newer version 1.3.40 which I would use to test. It would be very helpful to have 3-4 of those files, Perhaps while receiving and recording nmea0183, nmea2k and signalK and so labeled ?

I'll make the other issue now.

rgleason, 2024-12-29:
The newer VDR_pi v1.3.40 that was using is found here (for various OS)
https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=1.3.40+tarball

sebastien-rosset, 2024-12-30:
> I am still having issues with VDR playing older vdr created files. I have tried a half dozen files and none of them really work.
I think we need a set of testing files of various types to use to test the capabilities of the newer plugin.
I will make such an issue and post about 6 files for testing. This issue will also have a link to the older version of the plugin 1.3.31 and it will be showing the tarballs so the appropriate one can be downloaded and "Import Plugin" used to compare the results.

Could you elaborate on what does not work? I've analyzed in #55

rgleason, 2025-01-01:
@Sebastien-Rosset  
The files are working much better now, actually all of them. I will just show you one of them. I am quite pleased about this!!
Thank you.  Happy New Year.

[image]

PR #53 (2024-12-22): Files with no timestamp data
Fix #52 

> When using an old recorded vdr file from Harmut with the new version 1.3.39
> I find that moving the time line does not really work very well.

fixed

> When I try to move the start it is not recognized.

fixed

> I will show a few screenshots and upload a zipped text file of an older "Harmut" file recording so you can see what happens.

Date truncation fixed in replay window.

List of changes:
1. If the user is replaying a file that has no timestamp data, the slider should progress as a ratio of the current line number to the total number of lines. I.e [... cut]
closed 2025-01-01 by rgleason