rgleason/vdr_pi #54: Reminder: Android support

rgleason, 2024-12-19:
https://github.com/OpenCPN/OpenCPN/pull/4265#issuecomment-2549748911

sebastien-rosset wrote
I've been able to use the plugin API version 1.19 with GetListener. It works on Linux, MacOS and compiles on Windows (haven't tested), but the Android build fails. It looks like the Android library does not include the GetListener function.

bdbcat wrote:
Correct. Android production includes API118.

sebastien-rosset, 2024-12-19:
Now the Android build succeeds, but I don't know how to test.

rgleason, 2024-12-19:
Sebastien, this is a bit of a balancing act testing for Android. I don't do it often.
I have an android cell phone.   I get the necessary tarball from cloudsmith most likely the vdr-prod repository or if it has not been built for production it will be in vdr-alpha or vdr-beta   

Anyway this is the path to download the necessary android file (including a search to whittle it down) 

https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=1.3.39.0+android+tarball

Use the armhf or arm64 versions depending on the device.
Once downloaded, somehow you have to get it into your phone. I usually send as an attachment or connect to my laptop and copy over the wire.

Then what I do is simply import the plugin tarball.  This can be a bit of a challenge as Android and the Interface don't make it really easy.

I will try it too.

rgleason, 2024-12-19:
I have to copy the amr64  tarball file to 
PC\FREDERICK's Galaxy A51 5G\Internal storage\Android\data\org.opencpn.opencpn\files\plugins

Using Windows File Explorer

Then Import Plugin Tarball can access it.

rgleason, 2024-12-19:
I am not having much success. the arm64  tarball will load, without errors but it does not resolve and appear as a plugin.  This is the 1.3.39 version so perhaps we should be using a more recent version?

I also found there was no way to remove the two other vdr plugins  from the options.  I guess that is determined by the metadata file.

rgleason, 2024-12-20:
I have just realized that without VDR working in Android, we are basically unable to test other plugins  (weather_routing, autotrackraymarine, pypilot, tactics, dashboard_tactics, watchdog, etc.   We've just been relying on it working across OS and testing Mac, Linux and Windows.

sebastien-rosset, 2024-12-20:
I'm at an even more basic stage to try to setup the build environment for Android. The build instructions and very old, so I'm trying to update them, but I'm not close to having working instructions.

https://github.com/sebastien-rosset/OpenCPN/blob/android_instructions/buildandroid/README.md

rgleason, 2025-01-05:
Dave wrote:

> I will focus in on VDR for Android.
> Unfortunately, there is no easy way to build plugin locally for Android without spinning up the complete Android dev environment.  Best solution for a casual user is to use the CCI/Cloudsmith approach, and read the runtime logfile carefully.
>

rgleason, 2025-02-04:
I am going to close this, we've gotten over most of the android hurdles for this PI.  Using the CI build environment is the easiest way, but hampers fixing Android bugs quickly of course.

closed 2025-02-04 by rgleason