rgleason/vdr_pi #55: Testing Files

rgleason, 2024-12-29:
Using [VDR_pi 1.3.31 Tarballs](https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=1.3.31+tarball) which is the older version.
Some screenshots playing these files with Dashboard turned on.

[image]

[image]

[image]

[image]

[image]

These are the files I used which work with older VDR 1.3.31  They are generally standby files that I use when doing testing for something, although I have a whole messy directory of nmea0183 files testing various things ATON, AIS, Meteo, etc.
They are compressed and need to be unzipped into .txt files and then "Played" by VDR.

[Hakefjord-Sweden-1m.zip](https://github.com/user-attachments/files/18271140/Hakefjord-Sweden-1m.zip)

[Hartmut-AN-Markermeer-Wind-AIS-3m.zip](https://github.com/user-attachments/files/18271141/Hartmut-AN-Markermeer-Wind-AIS-3m.zip)

[NavMonPC-Hartmut.zip](https://github.com/user-attachments/files/18271142/NavMonPC-Hartmut.zip)

[PacCupStart.zip](https://github.com/user-attachments/files/18271143/PacCupStart.zip)

[Race-AIS-Sart-10m.zip](https://github.com/user-attachments/files/18271146/Race-AIS-Sart-10m.zip)

[Tactics-sample1-12m.zip](https://github.com/user-attachments/files/18271148/Tactics-sample1-12m.zip)

[Tactics-sample2-5m.zip](https://github.com/user-attachments/files/18271149/Tactics-sample2-5m.zip)

I hope this helps.

sebastien-rosset, 2024-12-30:
Thanks for all these test files! I will test with them.

sebastien-rosset, 2024-12-30:
The first file I tried is `PacCupStart.txt`, which works for me.

[image]

One noticeable characteristic of this file is that the timestamps in the RMC sentences are always rounded to the 10 seconds, and there are up to 10 consecutive RMC messages with the same timestamps, such as:

```
$ECRMC,194510,A,3749.012,N,12227.024,W,8.4,324.0,140708,015.0,E,A*1F
$ECRMC,194510,A,3749.017,N,12227.027,W,8.5,324.0,140708,015.0,E,A*18
$ECRMC,194510,A,3749.021,N,12227.030,W,8.7,326.0,140708,015.0,E,A*1B
$ECRMC,194510,A,3749.025,N,12227.033,W,8.8,328.0,140708,015.0,E,A*1D
```

sebastien-rosset, 2024-12-30:
In `Tactics-sample1-12m.txt`, I see the timestamps have milliseconds precision, e.g. `085349.800` is time in UTC  (08:53:49.800)

```
$GPRMC,085349.800,A,4503.7389,N,01338.8654,E,3.33,185.16,060516,,,D*6D
```

sebastien-rosset, 2024-12-30:
I'm able to replay `Tactics-sample2-5m.txt` successfully.

sebastien-rosset, 2024-12-30:
`NavMonPC-Hartmut.txt` does not seem to be a valid NMEA VDR recording. This appears to be a configuration file for "NavMonPc" (version 1.18).

sebastien-rosset, 2024-12-30:
The `Hartmut-AN-Markermeer-Wind-AIS-3m.txt` VDR file contains significant timestamp jump backwards. Previous ts: 2012-10-20T18:06:43.870Z, Current ts: 2012-10-20T17:35:15.000Z
Looking into it.

sebastien-rosset, 2024-12-30:
@rgleason , can you clarify which VDR you used when replaying and took the screenshots? The file names are truncated in the VDR replay window. Also, what are the actual symptoms? I do see the screenshots, but it's not very obvious what the problem is.

sebastien-rosset, 2024-12-30:
In `Hartmut-AN-Markermeer-Wind-AIS-3m.txt`, I've noticed there two sources for the `RMC` sentence, e.g.:
 
```
$AIRMC,093217.20,A,5229.5634,N,00505.7593,E,4.58,39.0,201012,0.7,W,D*35
$IIRMC,093216,A,5229.566,N,00505.758,E,04.6,037,201012,00,E,A*06
```

The `AI` source is not in sync with the `II` source (the II source is about 1.5 second behind `AI`.

sebastien-rosset, 2024-12-30:
When I try to install https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/detail/raw/VDR-v1.3.40-darwin-wx32-universal-10-tarball/v1.3.40/ on Macbook Pro 2.6 GHz 6-Core Intel Core i7 sequoi 1.15.1, the plugin does not work. The playback window is black

[image]

Clicking "preferences" cause segfault.

Same with version 1.37.1:

[image]

rgleason, 2024-12-30:
Sebastien, 
I see that you found the timestamps to have various anomalies, not sure why.  I do know they are quite old files, from earlier versions of OpenCPN and VDR.  Most of them were recorded with an earlier version of VDR, but some perhaps not, (NavMonPC).

I think I did not switch the Timestamp in Preferences and that was part of my problem running these files.
Should I try version 1.3.40 again?

Can you make a new separate issue on my VDR repo for Bug: MacOS plugin failure?
And add  @leamas, @bdbcat at the top?  
We seem to be having trouble with MacOS plugins for both Shipdriver Template (this plugin)  and the Testplugin Template.
That would be very helpful. Thanks

sebastien-rosset, 2024-12-30:
> Can you make a new separate issue on my VDR repo for Bug: MacOS plugin failure?

I've opened #57

sebastien-rosset, 2024-12-30:
> I see that you found the timestamps to have various anomalies, not sure why. I do know they are quite old files, from earlier versions of OpenCPN and VDR. Most of them were recorded with an earlier version of VDR, but some perhaps not, (NavMonPC).

It's not very surprising that some files may contain timestamps with some anomalies. For example, there may be two instruments reporting time, but the instruments are not in sync with each other. Or the user changes the time while VDR is in progress, or the user manually concatenates different VDR files into one file

I've been trying to handle these various scenarios by adding a tolerance factor. Also, if the timestamps go significantly backward at least once in the file, then replay should still work. In that case, the strategy is to ignore the timestamps and just replay the sentences. In the future, there could be a further enhancement where the plugin first identity a set of segments that have monotonically increasing timestamps. Then replay could be done separately for each segment.

In short, the VDR plugin work should work even on older VDR file. So feel free to report issues.

> I think I did not switch the Timestamp in Preferences and that was part of my problem running these files.
Should I try version 1.3.40 again?

> Also do you have some nmea2000 files I could run? Thank you for your great programming!!!

I don't have such files directly from my boat; I currently record NMEA 0183 because my B&J chartplotter emits NMEA0183 over TCP. I do intend to buy a wifi adapter that will be able to send NMEA 2000.
That being said, for ease of development, I also have a NMEA simulator, so I can generate arbitrary VDR files or directly feed the data into a OpenCPN connection. I have a few bugs to fix for NMEA 2000, but once this is done, I can send you some data. https://github.com/sebastien-rosset/nmea-simulator

rgleason, 2024-12-31:
Sebastien   I retested these files with VDR 1.3.40 and I confirm that the two files that work the best right now are PacCupStart and Race-AIS-Start.  Will test your newer version when it is finished.

sebastien-rosset, 2024-12-31:
> Sebastien I retested these files with VDR 1.3.40 and I confirm that the two files that work the best right now are PacCupStart and Race-AIS-Start. Will test your newer version when it is finished.

For the files that don't work, can you describe what specifically is not working? That will help to identify and fix specific problems.

rgleason, 2024-12-31:
Ok I will. I was using 1.3.40 a second time, the other files would start up turn red and then just sit there with no change in data displayed by Dashboard, then after awhile the boat would change color to black.  Moving the location in the file did not seem to help, and it always did that.  I tried slowing down the speed and speeding up, with no change.  I also had all kinds of AIS warnings popping up and I thought the bus might be stuffed from trying to run previous files, so I restarted Opencpn and turned off all AIS.
No change the boat turned red and didnt move then turned black.
https://github.com/rgleason/vdr_pi/pull/56

rgleason, 2025-01-01:
So people can find the test files.

rgleason, 2025-01-01:
These files are working now. Leaving this Issue up so people can find the test files (for the time being)

[image]

sebastien-rosset, 2025-02-04:
I've added these test files under `test/testdata` and used them in unit tests.

rgleason, 2025-02-04:
Great, I guess I should close this?

sebastien-rosset, 2025-02-04:
> Great, I guess I should close this?

Yes, I think so. If there are additional issues in existing files or new files, we can add them to the `testdata` directory and add unit tests for them.

closed 2025-02-04 by rgleason

rgleason, 2025-02-05:
Sebastien,  Thank you for your efforts to improve VDR.  This is remarkably better now because of all that you have done.

I tried all the files and they all work except for the NavMonPC-hartmut.txt  and that must be because it is too many sources and out of sync   or perhaps I am not using the right settings.

I've done some testing with Play, have not done much with recording though.  I've made a PR to release and publish. 
Thank you again!

rgleason, 2025-02-05:
Sorry, I forgot that!!!

PR #56 (2024-12-31): Fix sub-second parsing in RMC and ZDA sentences
Fix #55 

This is WIP.

## Issue 1: Parsing Sub-seconds in RMC and ZDA sentences

In `Tactics-sample1-12m.txt`, timestamps have milliseconds precision, e.g. `085349.800` is time in UTC  (08:53:49.800), which is supported by the NMEA standard. For example:

```
$GPRMC,085349.800,A,4503.7389,N,01338.8654,E,3.33,185.16,060516,,,D*6D
```

The code was incorrectly parsing sub-seconds.

## Issue 2: Log timestamps when time is not monotonically increasing.

Logging the timestamps helps to troubleshoot discontinuities, i.e. when the timestamps move backw [... cut]