rgleason/vdr_pi #57: VDR plugin on MacOS with Intel processor does not work

sebastien-rosset, 2024-12-30:
# Summary

The VDR plugin can be installed on a Macbook Pro with Intel processor. However:
1. Opening the VDR replay window shows a panel with uniformly black content.
2. Trying to click "Preferences" causes segmentation fault.

# Steps to reproduce

1. Run OpenCPN on a macbook pro with Intel processor, Sequoia 1.15.1
2. Download the VDR plugin from:
https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/detail/raw/VDR-v1.3.40-darwin-wx32-universal-10-tarball/v1.3.40/
3. Install the VDR plugin.
4. Click "Preferences" and notice OpenCPN crashes with a segmentation fault.
5. Restart OpenCPN. Click the VDR replay window. Notice a floating window is displayed, but the content of the window is entirely black.

jongough, 2024-12-31:
I just installed this plugin, 1.3.37.0, on macOS ventura -13 which is running as a VM guest on Ubuntu 24.04. I can use both the record and play buttons (although I have no GPS input so there is no data generated). In the 'settings/plugins' I can click on the VDR and get the opportunity to Reinstall/Uninstall, but the preferences are greyed out, so I cannot get to them to test your situation.

rgleason, 2024-12-31:
Jon, there is a VDR 1.3.40 and you can test with these files 
https://github.com/rgleason/vdr_pi/issues/55

Thanks.
Sebastien has improved 1.3.40 and tested 
see https://github.com/rgleason/vdr_pi/issues/55#issuecomment-2564955763
see https://github.com/rgleason/vdr_pi/pull/56
https://github.com/rgleason/vdr_pi/pull/56

but it is in draft form and will be merged at some point, when I will test too.

sebastien-rosset, 2024-12-31:
Personally, I'm going to upgrade my mac soon to Apple silicon, so this issue won't be relevant to me anymore. But still, I'm curious why there are problems.

jongough, 2024-12-31:
I have now tried 1.3.40 on the ventura-13 virtual mac and it seems to work OK. 

It would be nice to be able to select the alpha or beta repository and do a standard download/install process rather than downloading the tarball. Not being a mac user I find the 'Finder' confusing as it will not show the directories properly and I have to keep going to a Terminal and using the command prompt to do anything with testing and setup. Is there a chance this plugin will use the beta process and upload the requisite files to the plugins git project?

rgleason, 2024-12-31:
Jon, I can't answer your question. Perhaps Rasbats could advise?
Sorry the Mac Finder is hard to use.
**Thank you for this report!!**
We've accepted Sebastien's most recent improvements and will be pushing it to plugins. So that should help.
It will be v1.3.41

rgleason, 2025-01-01:
@sebastien-rosset   Can this be closed now? Thanks.

sebastien-rosset, 2025-01-01:
sure.

rgleason, 2025-01-01:
https://github.com/rgleason/vdr_pi/issues/64

closed 2025-01-01 by rgleason