rgleason/vdr_pi #58: RFE Menu Icons too small on High DPI Screens

rgleason, 2024-12-31:
Also on my High DPI Display 3840 x 2400 I think with Windows "Scale" at 250% which is "Recommended" the icons in the menu are pretty darn small. Will make separate RFE for this too.

[image]

sebastien-rosset, 2024-12-31:
Do you mean the "play/pause/stop" button in the VDR replay.
Also, I see the time is still truncated. This is a problem that needs to be fixed.

rgleason, 2024-12-31:
Yes, please.

rgleason, 2025-01-01:
Almost a duplicate Issue  https://github.com/rgleason/vdr_pi/issues/62

rgleason, 2025-01-01:
https://github.com/rgleason/vdr_pi/pull/63

rgleason, 2025-01-01:
Version 1.3.42 is much better, but:
1. Icons need to be bigger for High DPI.
2. Can the max speed of the speed slider be increased so that a file can be run considerably faster?

[image]

sebastien-rosset, 2025-01-01:
I'll work on two PRs, one for bigger icons, and one for increasing the speed.

rgleason, 2025-01-01:
Also please see the next Issue about using it on Android.  This is so close to really working well it is fantastic.
Summary: On android, the menu can't be moved and the icons are also too small.  Otherwise, I was able to play files and it works well!

sebastien-rosset, 2025-01-01:
For Android, I need step-by-step, reproducible instructions to build OpenCPN and the plugin locally.

I've started here: https://github.com/OpenCPN/OpenCPN/pull/4272, but I see there are several tricky parts that are being discussed here: https://github.com/OpenCPN/OpenCPN/issues/2081

rgleason, 2025-01-01:
@bdbcat  Can you advise Sebastien how to build locally for Android?  This VDR_pi is so close to being usable it is awesome.

rgleason, 2025-01-05:
VDR v1.3.44 is much better. There are a couple of things that could be adjusted for better control.

1. Speed control. I think something like 200x's speed, rather 1000x's would be better for practical reasons, but I can imagine if there is a huge file, someone might want to skip through it, but for testing 200x's would be better because it would give us better control on the slider.

2. I can't seem to get [two of the test files ](https://github.com/rgleason/vdr_pi/issues/55)to work.  

- Hakefjord-Sweden-1m.txt,   
- Hartmut-AN-Markermeer-Wind-AIS-3m.txt
These are time and date challenged recordings. They do not seem to run at all anymore.
I did not try NavMonPC-Hartmut
The others seem to work nicely.

These two files seem to get a black square that when you hoover it says "End of file" and the bottom bottom of the menu shows "Time stamps are not chronological"  When I click on the black square to run the file the file runs right out to the end without showing anything.  See the snapshots below.  These files would actually run in previous versions.

This is using Windows 11 with High DPI Screen, Also the icons could be a little bigger.
 
[image]

[image]

bdbcat, 2025-01-05:
This might be thought of as an RFE....
Would occasionally be useful to have the VDR playback proceed in "real-time".  That is, logic will parse the incoming NMEA sentences and extract the timestamp.  Then throttle the output such that the delta-time between messages matches the recorded timestamp delta-time.
This function would be useful for debugging certain types of downstream errors.
Thoughts?

rgleason, 2025-01-05:
@sebastien-rosset   Perhaps we should we make an RFE for this suggestion?

sebastien-rosset, 2025-01-05:
> This might be thought of as an RFE....
> Would occasionally be useful to have the VDR playback proceed in "real-time".  That is, logic will parse the incoming NMEA sentences and extract the [...]

This is what I have already implemented. For - CSV recording, it's easy because every sentence has a timestamp. For raw NMEA0183, the code parses RMC and ZDA sentences and extract date/time. When you play at 1x speed, the replay is in "real time".

If there are time discontinuities, such as time going backwards, the plugin falls back to reading line by line 

I do have further enhancements in mind:
1. Parse more NMEA sentence types that have date and/or time.
2. Identify time sources with unique (talker ID, sentence type, precision) tuple.
3. Identify continuous time segments that have at least one time source in chronological order with major discontinuities, i.e ltime is not going backwards, or significant jump forward.
4. Identify "primary" time source, ie "best" time source.
5. Interpolate time to assign timestamp to sentences that don't have a timestamp.
6. Expose time segments in the VDR control panel.

I have done the above in a python package, I'll try to port to the VDR plugin when I have time.

sebastien-rosset, 2025-01-05:
> VDR v1.3.44 is much better. There are a couple of things that could be adjusted for better control.
>
> 1. Speed control. I think something like 200x's speed, rather 1000x's would be better for [...]

You should be able to move the speed slider at 200x, or any value between 1x (real time) and 1000x.
I was contemplating implementing a log scale for the slider and enable even higher replay speeds. Then I thought it would be nice to first read all the sentences that have GNSS location and have the option to immediately display a track even before starting the replay. That would give an overview of the file content.

> 2. I can't seem to get [two of the test files ](https://github.com/rgleason/vdr_pi/issues/55)to work.
>
> - Hakefjord-Sweden-1m.txt,
> - Hartmut-AN-Markermeer-Wind-AIS-3m.txt
> These are time and date [...]

I have analyzed more files and concluded we need the following enhancements: 
1. Identify unique time sources within the file. Use (talker ID, sentence type, precision) tuple.
2. Identify primary time source. 
3. Identify continuous time segments, ie segments that have primary timestamps in chronological order.

That's because some recordings have multiple talkers IDs generating time stamps, and sometimes these talker IDs don't have time synchronization. That causes time stamps to not be in chronological order when reading the file linearly. 

Have also found some files where two instruments have the same talker ID.

bdbcat, 2025-01-05:
sebastien....
Thanks for the info.  This is all a great improvement. I will give it a try.

sebastien-rosset, 2025-01-06:
I want to point out right now nmea 2000 can be recorded but not replayed. I think ideally there would be multiple replay formats: TCP, UDP... And also support for different frame encapsulation, ie Arctic sense, yd...

rgleason, 2025-01-07:
VDR 1.3.53.0 Testing.  Windows11 
1. The installation problem is fixed. After update of catalog.  Not necessary to "Uninstall" and then "Install".  Just select "Reinstall" and the new version is listed.

The test files are located here https://github.com/rgleason/vdr_pi/issues/55#issue-2762437447

3. These files run very nicely, with interruptions, sliding both file location and speed while playing, etc.
- Race-AIS-SART-10m.txt
- PACcup-Start.txt
- Tactics-Sample1-10m.txt
- Tactics-Sample2-5m.txt

4. Some of the Nmea files do not work and are as described [here](https://github.com/rgleason/vdr_pi/issues/58#issuecomment-2571694483)  
-  Hakefjord-Sweden-1m.txt,
-  Hartmut-AN-Markermeer-Wind-AIS-3m.txt
-  NavMonPC-Hartmut.txt
-  These are time and date challenged recordings. They do not seem to run at all anymore.  The others seem to work nicely.
These files are not showing anything and do not play. Should we just drop them or have a message?

5. Icons still are not big enough for High DPI screens and  Android (finger size please). 
- https://github.com/rgleason/vdr_pi/issues/58
- Android https://github.com/rgleason/vdr_pi/issues/66#issuecomment-2571687172 (I need to check this again)

6. Android VDR Menu is frozen and cannot be moved with a finger.
-  Android https://github.com/rgleason/vdr_pi/issues/66#issuecomment-2571687172  (I need to check this again)

rgleason, 2025-01-13:
Sebastien, Can you fix this problem?
Also should we be selecting anything in Preferences?
[image]

rgleason, 2025-02-05:
I think this can be closed now, as Sebastien has done a nice job.  We can open new issues later if there are any.

closed 2025-02-05 by rgleason