rgleason/vdr_pi #66: RFE Android small fixes

rgleason, 2025-01-01:
VDR_pi 1.3.43.0  https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=android+tarball
VDR-Test-Files Zip: https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=VDR-test-files.zip

1. The menu does not seem to move. (Perhaps I have difficulty selecting it?
2. The menu icons are also too small (a different scale than High res DPI I think)

[image]
[VDR-test-files.zip](https://github.com/user-attachments/files/18286058/VDR-test-files.zip)

rgleason, 2025-01-05:
VDR_pi 1.3.44.0  update the plugin catalog. Hit Uninstall and then install v1.3.44
VDR-Test-Files Zip: https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=VDR-test-files.zip

See Importing Android Plugin Tarballs https://github.com/rgleason/vdr_pi/issues/64

    The menu still does not seem to move. Does anyone else have success moving it?
    The menu icons are also too small (a different scale than High res DPI I think)?

   
[image]

rgleason, 2025-01-05:
Dave wrote:

> I will focus in on VDR for Android.
> Unfortunately, there is no easy way to build plugin locally for Android without spinning up the complete Android dev environment.  Best solution for a casual user is to use the CCI/Cloudsmith approach, and read the runtime logfile carefully.

bdbcat, 2025-01-07:
I have pushed a PR containing a few small changes to improve the UI on Android.
Please merge and comment.
Thanks

rgleason, 2025-01-07:
Dave, your VDR 1.3.53 is much better. I will post some screenshots when I get them downloaded.
1. Installation works well, the only problem is it shows an Update with the wrong version number, but when you select that you are given the chance to select the new version.
2. Control Icons are bigger and much better for controlling.
3. The Race-AIS-Sart.txt file works!

These are now on my wish list:
1.  Make the two slider handles bigger so we can select them with a finger.
2. Allow the menu to be moved out of the way or in another place.

rgleason, 2025-02-04:
We had double versions in the master plugin catalog,  Limiting to one version solved a few of these issues.

1. One possible remaining issue is the size of the slider icon is too small to move with a finger.
3. Preferences button to Disable Sart, etc. which becomes an issue when playing some files.

rgleason, 2025-02-13:
The menus are accessible and work ok.
1. The small drag icon at the end of the speed line and file location line could be bigger in Android as they are too small now.
2. Dialog box is stuck in upper left next to first row of PI icons.  It does not seem to move.

[image]

bdbcat, 2025-02-13:
I would publish as is.  The current Prod offering is frankly not usable at all on Android.
Leave the Issue open, so that the author can address in due course.

bdbcat, 2025-02-13:
For posterity, I would add two more issues:
1.  Should convert buttons to wxBitmapButton model, with SVG icons.
2.  I don't think that the plugin has been fully localized yet.  Upload sources to crowdin, take submissions, and merge to CCI build.

rgleason, 2025-02-13:
This is actually great, we now have VDR working on android!
I didn't realize icons were not svg, I think the toolbar icons are svg, but perhaps not the others. I will check the code and try to convert them.
So we need to localize the plugin, I'll ask Hakan.
... and publish in present form anyway, with updates later.

Note:  Using the cellphone usb connection to PC, with Windows file manager, and copied the VDR test files to 

This PC\.. Galaxy A51 5G\Internal storage\Android\data\org.opencpn.opencpn\files\plugins\vdr_pi\data\Test

From there is was easy to test.

rgleason, 2025-02-13:
I was actually testing VDR 1.3.60, and not VDR 1.3.61.0   I apologize.  I thought we had published 1.3.61 but I will do it now.

rgleason, 2025-02-13:
I sent Sebastien an invite to be collaborator. 
I also made a PR to publish 1.3.61  https://github.com/OpenCPN/plugins/pull/1244

rgleason, 2025-02-14:
From VDR 1.3.62.0 on a  Samsung Galaxy A51 features a 6.5-inch Super AMOLED display with a resolution of 2400 x 1080 pixels.
1. The icons in Dialog got a little too small, and the drag icons are definitely too small.
2. The dialog is stuck in the upper left corner.
3. Sorry about this, but I am looking at this as though it is the first plugin that is actually working properly in android, using it as an example.

It is playing one of the files nicely.

[image]
[image]