rgleason/vdr_pi #72: RFE: Real time playback.

rgleason, 2025-01-07:
Would occasionally be useful to have the VDR playback proceed in "real-time". That is, logic will parse the incoming NMEA sentences and extract the timestamp. Then throttle the output such that the delta-time between messages matches the recorded timestamp delta-time.
This function would be useful for debugging certain types of downstream errors.
Thoughts?

sebastien-rosset, 2025-02-04:
Everything above has been implemented in #88 , except for:
> 5. Interpolate time to assign timestamp to sentences that don't have a timestamp.
6. Expose time segments in the VDR control panel.

rgleason, 2025-02-04:
Perhaps we can try it out and if people think those have use, we'll make a new issue.  So let's close this, can be reopened.

closed 2025-02-04 by rgleason