rgleason/vdr_pi #74: RFE: Certain old Nmea0183 files do not work

rgleason, 2025-01-07:
Rick
> I can't seem to get https://github.com/rgleason/vdr_pi/issues/55to work.
> 
>         Hakefjord-Sweden-1m.txt,
>         Hartmut-AN-Markermeer-Wind-AIS-3m.txt
>         These are time and date challenged recordings. They do not seem to run at all anymore.
>         I did not try NavMonPC-Hartmut
>         The others seem to work nicely.

Sebastien

> I have analyzed more files and concluded we need the following enhancements:
> 
>     Identify unique time sources within the file. Use (talker ID, sentence type, precision) tuple.
>     Identify primary time source.
>     Identify continuous time segments, ie segments that have primary timestamps in chronological order.
> 
> That's because some recordings have multiple talkers IDs generating time stamps, and sometimes these talker IDs don't have time synchronization. That causes time stamps to not be in chronological order when reading the file linearly.
> 
> Have also found some files where two instruments have the same talker ID.

rgleason, 2025-01-25:
Dave wrote:

> I attach an existing VDR recording that will not play correctly in the new version.  I poked around in the code a bit, and find that the trouble is this:
> 
> In this file, some messages contain timestamps, and some do not. A not uncommon situation with multiple talkers.
> 
> This appears to confuse the scheduler, and has the result that the entire file is spit out at 1 msec interval, so completely unusable.  A regression.
> 
> Please take a look, and advise.
> 
> I think a reasonable solution in this scenario is to pace the output at the target rate of any sentences that contain timestamps, as encountered.
> 

[hakan.txt](https://github.com/user-attachments/files/18546291/hakan.txt)

NOTE:  If these two were fixed, we can publish it 
https://github.com/rgleason/vdr_pi/issues/74
https://github.com/rgleason/vdr_pi/issues/82

sebastien-rosset, 2025-02-04:
I've improved the timestamp parsing in #88.

rgleason, 2025-02-04:
Thank you Sebastien!

closed 2025-02-04 by rgleason