rgleason/vdr_pi #77: Bug Flatpak aarch64 22.08  Issues

rgleason, 2025-01-09:
Raspberry Pi5 Bookwormm,  OpenCPN 10.5.2 flatpack
flatpak-aarch64-22.08-flatpak

Manfred reports
https://www.cruisersforum.com/forums/f134/plugin-testing-all-operating-systems-needed-291446.html#post3963343

VDR Recorder 1.3.53 do not see recorded files in DataManager

Playback icons are not visible and do not respond to mouse clicks, can't replay file
[image]

rgleason, 2025-02-05:
I wonder if this still exists? I guess I'll leave it open and see what Manfred reports.

sebastien-rosset, 2025-02-05:
@rgleason , do you have any hint about how to troubleshoot this issue?

rgleason, 2025-02-05:
I think that maybe the data/icons were not copied into the tarball? We should look at the tarball and make sure.  The other thing it might be is a wronf path or not initailized, but the other versions seem to work.
I was wondering about this to.

rgleason, 2025-02-05:
I am going to ask Manfred about this.

sebastien-rosset, 2025-02-06:
The "icons" are actually UTF-8 characters, not SVG images. I'll try to figure out why they are not displayed.

[image]

rgleason, 2025-02-13:
Did this get fixed?  I don't have flatpak, but can ask users on the forum to check if needed.

sebastien-rosset, 2025-02-13:
I don't think this got fixed. Probably using SVG should fix. #96. I'll let you know when it's ready for merge.

PR #96 (2025-02-14): Use SVG images instead of UTF-8 characters
Use SVG images instead of UTF-8 characters. Hopefully this will fix #77. I'm still not sure how to test whether this will work on all platforms, so we may have to try and iterate if icon sizing issues are found.

[image]

# Additional changes

1. Remove `VDR_USE_SVG ` compile-time macro. This was enabled on all platforms, so we can remove the `#ifdef` `#else` statements 
2. Remove SVG that were hard-coded in the code. Instead, the SVG are being loaded from the package.
3. Remove old SVG icons that are no longer used.
4. Fix sizer WX alignment issues f [... cut]
closed 2025-02-14 by rgleason