rgleason/vdr_pi #80: NMEA 0183 speed detection it too specific

rosinbum, 2025-01-13:
The NMEA 0183 spec allows for many "talkers" to send messages. In the NMEA 0183 string like `$GPRMC,195936.00,A,3025.0869,N,09754.1536,W,7.8,151.5,130124,,E,A,C*6A<0x0D><0x0A>` the `GP` indicate the talker. ie the device sending the message. The `RMC` part designate what type of message it is. Therefore if the talker doesn't have the designation `GP` or `GN` the message is not recognized as type `RMC - GPS recommended minimum`.

Messages like `$YDRMC,195936.00,A,3025.0869,N,09754.1536,W,7.8,151.5,130124,,E,A,C*6A<0x0D><0x0A>` are just as valid with the `YD` talker. 

Further there are other non GPS speeds that may make better sense to use as doing 5 knots into a 5 knot current renders a GPS speed of 0 and would trigger a stop in recording. The `VHW` or `VTG` may be better candidates.

rgleason, 2025-01-13:
These are good points and they need some thought and discussion. I hope sebastien will join the discussion here too.  @sebastien-rosset

rosinbum, 2025-01-13:
I think as a stop gap eliminating the specific talker and accepting any `RMC` sentence regardless of talker is a good start. That's what #81 does. 

I agree that the larger issue of looking at alternative sources of speed do need some conversation.

sebastien-rosset, 2025-01-13:
I agree it is good to remove the talker id. As a further enhancement, it would be good to create a set of unique time sources based on (talker ID, sentence type, time precision) tuple. In principle, the (talker ID, sentence type) should be sufficient but I've seen VDR recordings where two instruments are emitting the same GPRMC sentence, and with slightly out-of-sync time and different number of decimals.
Once the set of time sources has been determined, it's possible to select a primary time source and then interpolate the time for all other sentences. This would help to handle the case when multiple instruments are sending timestamps that are not in chronological order because these instruments are not exactly in sync.
The issue is that if during replay, the plugin does not have a means to sequence sentences based on chronological order, then it wouldn't be able to determine the start/end time, and the current time being replayed.

sebastien-rosset, 2025-01-13:
I also agree with the suggested speed improvements. For now having a large enough "x minutes after speed goes below threshold" value should take care of the current against heading scenario, as it seems unlikely a boat would not move at all for an extended period of time unless it is at anchor or dock.

PR #81 (2025-01-13): Allow alternate talkers for NMEA0183 RMC sentences.
Check for the RMC type of NMEA sentence instead of the full sentence header that is talker specific.

This resolves #80
closed 2025-01-13 by rgleason