rgleason/vdr_pi #82: BUG VDR 1.3.53 Play Menu > Preferences and Play file do not work when "No File Loaded"

rgleason, 2025-01-17:
I believe this bug was also causing Plugin Menus to load very slowly. See https://github.com/OpenCPN/OpenCPN/issues/4314

After renaming my opencpn.ini and reinstalling all my charts etc., then updating plugin catalog and testing the Toolbar, it opened in 6.5 seconds instead of 19 seconds.

Then I tried VDR and found this issue.  Preferences and Play file do not work.
I believe there were some left over settings under VDR in opencpn.ini that were bogus and phantom, that were causing troubles for opening the Toolbar.

In any event VDR 1.3.53 does not work and should probably be removed from the catalog, unless this is some anomaly.

[image]

Hakansv, 2025-01-17:
I noticed today the VDR record does not add CRLF at each row end. So whatever is recorded it's only one, very long, row is created.
I had to manually add CRLF  to get it to play.

rgleason, 2025-01-17:
We may have to remove 1.3.53 which has major improvements, due to these bugs.

Hakansv, 2025-01-17:
My VDR has version 1.3.0 and is the only available in master catalog.

rgleason, 2025-01-17:
Weird, that is an old version. Did you update the catalog? It should have 1.3.53

Hakansv, 2025-01-17:
catalog master is updated, yes. (Win10)
1.3.30.3 on Rpi5 Bullseye

Hakansv, 2025-01-17:
Hmm.
That's the pluginmanager bug.
If I select "Reinstall" 1.3.31 and 1.3.53.0 is available. But no green up-arrow?
Should I try any of them?

Hakansv, 2025-01-17:
This is strange. VDR recording was working before and i obviously have an old version.
I'm on master branch. Could changes to the OCPN connections has something to do with the missing CRLF?

rgleason, 2025-01-17:
RE No Green Arrow for upgrade. I too did not have that, but when I initially installed 1.3.53 my cache was entirely cleaned out of weatherfax and I did see a green up arrow, so I thought that the problem was fixed.

I also think that this "No Green up arrow"  for Upgrade occurs after an Import Plugin of a tarball.

Hakansv, 2025-01-17:
> I also think that this "No Green up arrow" for Upgrade occurs after an Import Plugin of a tarball.

VDR I've never imported so not related here.

rgleason, 2025-01-17:
@sebastien  I am going to remove the 1.3.53 because it is not working. This will leave the older version.

Hakansv, 2025-01-17:
Agree about not working 1.3.53 on W10
Installed 1.3.31
When recording:
"SetNMEASentence()" $XXYYY does not include CRLF
however "SetNMEASentence()" does.
NMEA is written one row until interrupted by a AIS sentence, if any. 
All AIS sentences are though written including CRLF.

Hakansv, 2025-01-18:
Forget about the comment copied below. It's not correct.
I was using a GNSS puck that shows other obstacles.
When using a proper device all is fine with VDR recording. Sorry for the noise.
Håkan
> "SetNMEASentence()" $XXYYY does not include CRLF
> however "SetNMEASentence()" does.
> NMEA is written one row until interrupted by a AIS sentence, if any.
> All AIS sentences are though written including CRLF.

rgleason, 2025-01-18:
Sebastien,
I tried a novel approach. 
Using the older plugin, I loaded a nmea file and closed the plugin.
Then installed a downloaded tarball using Import Plugin Tarball.
Then VDR worked fine.
VDR just needs to handle when there is no nmea file entered.

rgleason, 2025-01-25:
If there is no startup file for VDR_pi already in place, (in opencpn.ini) then the plugin is essentially frozen and not able to open a file.

The way around this is to use an earlier VDR version and load any nmea file, close and then reinstall the new vdr PI.

Then after that the plugin works and can load other files, but initially it is clueless.

sebastien-rosset, 2025-02-04:
Catching up to this thread.

> I noticed today the VDR record does not add CRLF at each row end. So whatever recorded it's only one, very long, row created.
I had to manually add CRLF to get it to play.

Do you mean that when writing NMEA0183 sentences to the VDR file (in raw NMEA format), there are no CRLF characters?

Hakansv, 2025-02-04:
As noticed 2 weeks ago: 

> Forget about the comment copied below. It's not correct.
> I was using a GNSS puck that shows other obstacles.

Sorry again for the noise. If it's up to me, we can close.

rgleason, 2025-02-04:
Hakan, your post was very useful, and got us thinking about it. Thanks. Closing.

closed 2025-02-04 by rgleason