rgleason/vdr_pi #86: Bug: Stop after speed drops below functionality doesn't work speed stops being sent.

rosinbum, 2025-01-20:
The stop recording when speed drops below X for Y amount of time only checks X and Y if there speed sent into the system. If for some reason no speeds are transmitted then VDR never stops recording. This can happen quite easily. For example if the boats instruments are turned off once it is anchored/docked and it hasn't been Y amount of time.

It would be better to use a timeout style check that updates the timeout every time a speed above X is detected.

rgleason, 2025-01-25:
@rosinbum   Please elaborate

1. To enable greater understanding of exactly what is happening or not happening. (Step by step)
2. Then describe your suggested solution.

Thanks.

rosinbum, 2025-01-29:
The code which checks to see if auto recording should be stopped is in the `CheckAutoRecording` function. That function is only called when there is a speed message.

For n2kmessages:
https://github.com/rgleason/vdr_pi/blob/1d7e10d77cf2cd8c290413d995f1ce2da7870fea/src/vdr_pi.cpp#L330

and for nmea0183:
https://github.com/rgleason/vdr_pi/blob/1d7e10d77cf2cd8c290413d995f1ce2da7870fea/src/vdr_pi.cpp#L411

The `CheckAutoRecording` function is only called if there is a speed message. So what happens if speed messages stop being received? The check to stop auto recording is not made. 

My particular use case is that I have openCPN running on my laptop connected the boat instrument wifi. If I turn the boat instruments off the boats wifi turns off as well. Then openCPN stops receiving NMEA messages. No messages -> no speed updates -> no checkAutoRecording call -> Recording doesn't stop. 

The proposed solution is to use a timer that stops recording after it has reached the timeout as configured. Each time there is a speed message, reset the timer. When the timer reaches the configured limit you know that there has not been a speed message in the configured interval and then stop recording.

sebastien-rosset, 2025-02-04:
Sure, we could enhance the plugin to add a message timeout.
Also, can you elaborate on what are "and other problems" as mentioned in the issue title?

sebastien-rosset, 2025-02-04:
> The stop recording when speed drops below X for Y amount of time only checks X and Y if there speed sent into the system. If for some reason no speeds are transmitted then VDR never stops recording. [...]

If the instruments are shutdown, then presumably no NMEA sentences is produced, hence nothing is written to the file. Sure, the recording isn't paused, but nothing is written so the file isn't growing. This was by design.

Or are you saying the instruments are shutdown partially? I.e. some instruments continue to produce NMEA sentences, but the instrument that produces the `RMC` sentence was shutdown. In that case, we could consider enhancements to analyze other sentences such as GNSS coordinates. For example, we can calculate the speed from the coordinates, or detect if the vessel is within a radius of fixed coordinates.

sebastien-rosset, 2025-02-04:
> Sure, we could enhance the plugin to add a message timeout.

Actually, thinking about it more, it's pointless to pause recording if there are no incoming messages. In that case, no messages are written to disk. It makes more sense to enhance the plugin to trigger pause based on other sentence types.

sebastien-rosset, 2025-02-04:
Current Behavior:

1. Pauses recording when Speed Over Ground (SOG) from RMC sentences drops below threshold X for time period Y
2. Main limitation: SOG only reflects movement relative to ground, not actual vessel state
3. Edge case: SOG could be 0 while vessel is actively moving (e.g., SOG cancels out with current)

Potentially could improve with additional data sources:
1. RMC - Lat/Lon + SOG + Course Over Ground
2. GGA - High precision Lat/Lon + GPS quality indicators
3. GLL - Basic Lat/Lon
4. VTG - Course and Speed Over Ground
5. VHW - Water Speed and Heading (useful for detecting actual movement)

This is a non-trivial enhancement and there are pitfalls. In particular, there can be any arbitrary combination of sentences.

sebastien-rosset, 2025-02-04:
@rosinbum , can you elaborate on

> Stop Recording when 0 speed settings reached doesn't work

What does not work? The current implementation is to get the SOG from the RMC sentence:
1. If the SOG goes below a threshold for a configurable amount of time (2 minutes by default), then pause recording.
2. If the SOG goes above a threshold again, then resuming recording.

rgleason, 2025-02-04:
@sebastien The current implementation as you describe above appears to be reasonable to me.

sebastien-rosset, 2025-02-04:
One small enhancement is to improve the label to make it less ambiguous.

[image]

The plugin uses the Speed over Ground, not Speed over Water. I think one could argue for one vs the other, or even both, especially when there is a strong current. "Boat speed" is a bit ambiguous.

rgleason, 2025-02-04:
I like "speed over ground" as a clearer statement. I suppose it could be optional one or the other, but let's see how this works in practice.

sebastien-rosset, 2025-02-04:
I'll submit a PR for "speed over ground".

In principle, it would be great to auto-detect "at anchor". We could use a combination of many triggers, including even machine learning. But all of this seems quite overkill.

Start auto-recording immediately when (select zero or more):
1. Speed over ground exceeds xxx knots (from RMC/VTG)
2. Speed over water exceeds xxx knots (from VHW)
3. Position changes more than xxx nautical miles (from RMC/GGA/GLL)
4. Receiving anchor windlass/chain counter messages (NMEA 2000 PGN 128777)
5. Receiving any NMEA messages

Pause auto-recording when ALL selected conditions are met for xxx minutes:
1. Speed over ground below xxx knots (from RMC/VTG)
2. Speed over water below xxx knots (from VHW)
3. Position within xxx nautical miles of reference point (from RMC/GGA/GLL)
4. Anchor windlass status indicates deployed (if available via NMEA 2000)
5. Heading changes less than xxx degrees (suggests not actively sailing/motoring)
5. Not receiving specific message types:
   - RMC or VTG messages
   - VHW messages
   - Any NMEA messages
6. Position movement pattern matches anchor swing:
   - Movement follows arc pattern
   - Distance from center point varies with tide
   - Direction changes correlate with wind/tide changes

Additional Options:
- Average multiple readings before triggering state change
- Use GPS quality indicators (from GGA) to validate position

rgleason, 2025-02-04:
Wow, that is a lot of conditions and options.

rosinbum, 2025-02-04:
Wow, you guys are busy. One quick though is that it is possible for SOG messages to not be sent while other NMEA messages are. This may become even more true when there is a signalK implementation. So timeout may still be the appropriate solution. Either way thanks for all your work on this.

sebastien-rosset, 2025-02-05:
> One quick though is that it is possible for SOG messages to not be sent while other NMEA messages are. This may become even more true when there is a signalK implementation.

Why would it be more true for SignalK?

At some point I'll add support for SignalK recording, when api-19 is available for Android. When that happens, the user will be able to select which messages are recorded (any of NMEA0183, NMEA2000, SignalK). In addition, the speed is obtained from the applicable message type for that protocol. I.e. for NMEA 2000, the SOG can be obtained from PGN 129026, 129029; for SignalK SOG is obtained at path `navigation.speedOverGround`.

[image]

In principle, I agree there might be a stream of NME0183 messages without RMC.

rosinbum, 2025-02-05:
my understanding for signalk is that it is based on deltas and often/usually only sends out data after a protracted period of time of if there is a change to the value. If the speed is 0 for a long period of time it may not send a speed message. I believe there are ways to force a value to be sent at a min interval even if there is not a change but that is configured on a case by case basis.

https://signalk.org/specification/1.7.0/doc/data_model.html#delta-format

sebastien-rosset, 2025-02-06:
> my understanding for signalk is that it is based on deltas and often/usually only sends out data after a protracted period of time of if there is a change to the value. If the speed is 0 for a long [...]

The SignalK implementation has yet to be done in the VDR plugin, and it's quite likely additional logic will need to be added to handle the situation you describe.
Similarly NMEA 2000 replay requires custom logic for parsing PGN 129026 for SOG.
I'm planning to add option to replay over a network port (TCP, UDP) such that from the OpenCPN perspective, there would be no difference between a real connection and the replay connection (which might be useful when testing OpenCPN core functionality). Specifically for NMEA 2000, ideally the VDR plugin should support custom serialization variants (actisense serial, YD raw, etc).
But for NMEA 0183 (the current replay implementation), I'm not convinced with need a timeout mechanism.

rgleason, 2025-02-13:
Can we close this now rosinbum?  You are the OP and responsible for closing.  If there is some additional consideration later with signalK a new Issue can be opened. Ok?

closed 2025-02-16 by rosinbum