rgleason/vdr_pi #104: Badly sized icons on Windows

Hakansv, 2025-02-21:
The settings and load icons are not sized correctly.
See shots

Windows (Scaled 175%):
[image]

Rpi5:
[image]

rgleason, 2025-02-22:
Windows 11,  Opencpn 5.10.2, vdr 1.3.62
High DPI screen  Display Resolution: 3840x 2400   Scale:   250%  (Recommended)

[image]

If we can get this right, then we can fix the other plugins too.

Hakansv, 2025-02-22:
Compare Dashboard properties or Grib UI. They are fine for me.

rgleason, 2025-02-22:
> Hakan "Compare Dashboard properties or Grib UI. They are fine for me."

In Android   using Dashboard when I set for Standard Font size in Dashboard, its not at all legible.

[image]

rgleason, 2025-02-25:
I started this issue 4 days ago   I've incremented version to 1.3.63 and it is here now   https://cloudsmith.io/~opencpn/repos/vdr-prod/packages/?q=msvc  and this is the result with sebastien's changes.    VDR v1.3.63

[image]

So the issue is in the rendering of svg icons in windows and perhaps other platforms.
Will publish next.

rgleason, 2025-02-27:
These changes 
https://github.com/rgleason/vdr_pi/commit/c895a03b979f4862973b5cafbdf6b22f415f3f44

fixed High Resolution 3480x 2400 high dpi Windows screens.

[image]

rgleason, 2025-02-27:
Fixed by  [Adjust SVG icon sizes for scaled display on Windows](https://github.com/rgleason/vdr_pi/commit/c895a03b979f4862973b5cafbdf6b22f415f3f44)

closed 2025-02-27 by rgleason