rgleason/vdr_pi #107: Metadata: source is missing

leamas, 2025-04-21:
In current catalog the '<source>' entry for the vdr plugin is empty.  Makes it harder to find the repo maintaining the plugin.
For context, the first lines in each catalog entry:
```
    <plugin version="1">
    <name> VDR </name>
    <version> v1.3.65.0 </version>
    <release> 0 </release>
    <summary> Voyage Data Recorder </summary>
    <api-version> 1.18 </api-version>
    <open-source> yes </open-source>
    <author> Rick Gleason </author>
    <source>  </source>
```

rgleason, 2025-04-21:
What is missing?

```
<plugin version="1">
  <name> VDR </name>
  <version> v1.3.65.0 </version>
  <release> 0 </release>
  <summary> Voyage Data Recorder </summary>

  <api-version> 1.18 </api-version>
  <open-source> yes </open-source>
  <author> Rick Gleason </author>
  <source>  </source>
  <info-url> https://opencpn.org/OpenCPN/plugins/vdr.html </info-url>

  <description> A Voyage Data Recorder (VDR) to record and play NMEA files.
Save NMEA stream to a file.  Replay NMEA stream previously saved. Used to test plugins.
 </description>

  <target>android-arm64</target>
  <target-version>16</target-version>
  <target-arch>arm64</target-arch>
  <tarball-url> https://dl.cloudsmith.io/public/opencpn/vdr-prod/raw/names/VDR-v1.3.65.0-android-arm64-A64-16-tarball/versions/v1.3.65.0/VDR-v1.3.65.0_android-arm64-16-arm64.tar.gz </tarball-url>
  <tarball-checksum> 17d48ff41e5b691f53f38898838f84cfd48c5c313bedf077fbf1a53fe5d7e494 </tarball-checksum>
</plugin>

```

leamas, 2025-04-21:
The `<source>` tag is empty. Compare to for (surprise!) shipdriver:
```
  <name> ShipDriver </name>
    <version> v3.3.3 </version>
    <release> 0 </release>
    <summary> Simulate ship movements </summary>
    <api-version> 1.18 </api-version>
    <open-source> yes </open-source>
    <author> Mike Rossiter </author>
    <source> https://github.com/Rasbats/shipdriver_pi </source>   <-------------------
    [...]

```

leamas, 2025-04-21:
This is (should be) set in *Plugin.cmake* which should contain a line like `set(PKG_HOMEPAGE https://github.com/Rasbats/shipdriver_pi)`. Seems that this has been removed in vdr_pi cfor some reason

rgleason, 2025-04-21:
Thanks. Will fix.

rgleason, 2025-04-21:
It has

set(CPACK_PACKAGE_HOMEPAGE_URL https://github.com/rgleason/vdr_pi)

rgleason, 2025-04-21:
Changed and pushed a new version. Will publish to plugin master later.

rgleason, 2025-04-27:
Closing. Completed.

closed 2025-04-27 by rgleason