rgleason/vdr_pi #108: Data formats: VDR cannot read files created by current ocpn master

leamas, 2025-04-21:
Currently, the new Data Monitor in OpenCPN can create log files according to the spec in https://opencpn-manuals.github.io/main/vdr/log_format.html.  However, current VDR plugin seemingly cannot read these files as reported in https://github.com/OpenCPN/OpenCPN/discussions/4456#discussioncomment-12826494

It would be nice to confirm that both the core and the VDR plugin interprets the spec in the same way before the 5.12 release.

@sebastien-rosset

leamas, 2025-04-21:
Attaching a sample log from opencpn current  master

[data-monitor.csv.gz](https://github.com/user-attachments/files/19833629/data-monitor.csv.gz)

rgleason, 2025-05-04:
VDR and OpenCPN using different timestamps?

leamas, 2025-05-04:
No, there is a spec

TwoCanPlugIn, 2025-10-14:
The current VDR NMEA 2000 log format is incorrect.

The current output when recording NMEA 2000 looks like:
```
$PCDIN,130306,93130402FD01FF0200000000084DCF01FE4AFAFFFFF0
```

Which appears to be the bastard child of the Chetco Seasmart protocol and the Actisense RAW  protocol

The Seasmart Protocol actually looks like:
```
"$PCDIN,01FD02,000C8472,09,04C300487FF8FFFF*55";
```

Nonetheless, the supported format for the new loopback driver is:
```
NMEA2000 COM5 130306 93 13 02 02 fd 01 ff 03 ff ff ff ff 08 24 cf 01 fe 4a fa ff ff 55";
```

If VDR is going to provide an option to use the internal api (similar to the option for replaying NMEA 0183 logs), then it needs to support the same data format as the loopback driver.

Have a look at opencpn/opencpn#4387 for further details

closed 2025-11-29 by leamas